Jackysi/advancedtomato #31: Custom portal get lost afther flash

RobbyTown, 2015-01-01:
A few days ago i flash from 123 to 124. In version 123 i build my own guest portal page. Afther flash to 124 you get the fault guest html portal and all my own files are gone. I had a backup on my pc so it was fixt fast. 

All other settings about guest network dont change or get lost.

Jackysi, 2015-01-04:
I'm not sure that is a bug, you should write this to Shibby. This type of issues Shibby handles.

Jackysi, 2015-03-08:
I apologize for such long delay in response. http://linksysinfo.org/index.php
On that forum all the talk about Tomato goes on. AdvancedTomato is GUI modification of Tomato by Shibby mods. So he is the one doing such features and fixing them. Sorry

closed 2015-03-08 by Jackysi