Jackysi/advancedtomato #33: Port Forwarding > Basic - Weird Tab Order

windy-, 2015-01-25:
In the Port Forwarding > Basic page (#forward-basic.asp), if I type in a single port # (such as 100) in the Ext Ports field, the tab order works fine as it goes to Int Ports, Int Address, Description and finally Add+ button when I hit the tab key consecutively. However, as soon as I put in a range (such as 100-200) for the Ext Ports, the tab order becomes unpredictable. 

I have tested this on Asus RT-AC68U (2.5-124 K26ARM USB AIO-64K) and Asus-RT-N66U (2.5-124 K26AC USB AIO-64K).

Jackysi, 2015-03-08:
I apologize for such long delay in my response. I was extreme busy as you may imagine. I will look into issue!

Jackysi, 2017-11-14:
I have tested this 10x now on RT-N66U and AC32000. I can't find the problem.

closed 2017-11-14 by Jackysi