Jackysi/advancedtomato #34: Missing Router Name

windy-, 2015-01-25:
I don't know if you would consider this as a bug or feature request, but in shibby's tomato, you can always see the router name at the upper right corner of the page (see http://tomato.groov.pl/?page_id=31 for screenshots). However, it is missing in AdvancedTomato. I would imagine this is important for people who manage more than 1 router with the same theme. Here are some suggestions of placement of the router name (listed in no particular order):

- Upper right corner just like shibby's
- The white area next to the Toggle Collapsed Navigation button
- Near or replace AdvancedTomato logo in blue area
- Lower left corner near the version info

Jackysi, 2015-01-25:
Sorry but I disagree. I already added router name into the "System" box. If you click "System" you get router information which also shows which router it is. Also when you open Router page, overview states exact which router you have as well.

Sorry but I have to disagree and refuse this suggestion. Thank you anyway!

closed 2015-01-25 by Jackysi

windy-, 2015-01-25:
I respect your decision but I think you may have misunderstood my issue. I am talking about the router name, not the router model. If you look at http://tomato.groov.pl/wp-content/gallery/screenshots/bwlimiter.png, the router name is "groov.pl". My point is if I had two windows open, the only other way I could possibly differentiate between two routers of same model is by looking at the URL, which can be hidden with some web browsers.

Jackysi, 2015-01-26:
I understand, but window title starts with [ROUTER NAME] - PAGE is that not enough? lol