Jackysi/advancedtomato #36: "Error Occured" after configuring tinc

mikeodr, 2015-01-26:
After configuring tinc and starting the service the following error is posted any time I try and visit the tinc configuration page.

This persists even after router reboot.

* AP Model: Asus RT-N66U
* Error:

```javascript
TypeError: Cannot read property 'rows' of null
    at Object.TomatoGrid.getAllData (http://172.16.0.1/tomato.js:1522:51)
    at verifyFields (eval at  (http://172.16.0.1/js/jquery.min.js:2:4243), :163:16)
    at init (eval at  (http://172.16.0.1/js/jquery.min.js:2:4243), :384:1)
    at eval (eval at  (http://172.16.0.1/js/jquery.min.js:2:4243), :1:1)
    at eval (native)
    at Function.x.extend.globalEval (http://172.16.0.1/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://172.16.0.1/js/jquery.min.js:3:21416)
    at x.fn.extend.append (http://172.16.0.1/js/jquery.min.js:3:18985)
    at null. (http://172.16.0.1/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://172.16.0.1/js/jquery.min.js:2:5780)Cannot read property 'rows' of null
```

Jackysi, 2015-03-04:
I will check soon. I apologize for late response!

mikeodr, 2015-03-04:
No worries, non urgent for myself. Just ran across it.

Jackysi, 2015-03-05:
I think I fixed the issue. It should work in next release.

closed 2015-03-08 by Jackysi