Jackysi/advancedtomato #37: USB Custom Path, Logging IP Traffic/Bandwidth {does not work}

travisnj, 2015-02-18:
Your router brand & model: RT-AC56U
Interface version: 2.5-124 K26ARM USB AIO-64K

Screen shot or error details: 
EXT3 formatted USB stick mounted, will not log data/read data

How to reproduce the issue/error

1.) Format USB memory stick (mine: SanDisk U3 Cruzer Micro 4GB) EXT3, mount
2.) Custom configuration for all logging to /tmp/mnt/DATA/ (with or without trailing /) does not work

Jackysi, 2015-03-04:
This is strange. This works for me on R7000. I use an USB formatted drive via EXT3 as well.
In fact as funny as this is I use the same name for the drive as well!

Jackysi, 2015-03-08:
I realized this is specific issue to the device and it is not in my knowledge to fix it. Sorry. 
If you wish to have this problem resolved you will have to contact main developer Shibby who handles most of tasks like this. I am only a GUI developer sorry.

closed 2015-03-08 by Jackysi