Jackysi/advancedtomato #38: Default Splash Screen for Captivate Portal Loads to Blank White Page

r3n00, 2015-03-02:
Router Model: Linksys E4200v1
Tomato Version: 1.28.0000 MIPSR2-2.5-124 K26 USB AIO

The default splash screen (/tmp/splashd/splash.html) for the Captivate portal re-directs users to a blank white page when Captive Site Forwarding is enabled. Splash.html seems to reference style sheets that do not exists on the router 

Splash.html
-------------------------------------------------------------------------------------------------------------------------
 <title>$GatewayName</title>
                <link href="/css/bootstrap.min.css" rel="stylesheet">
                <link href="/css/bootstrap-responsive.min.css" rel="stylesheet">
                <link href="/css/tomato.css" rel="stylesheet">
                <link href="/css/elenor.css" rel="stylesheet">
---------------------------------------------------------------------------------------------------------------------------

Find output

-----------------------------------------------------------------------------------------------------------------------------
 root@home:/# find / -name "*.css"
/usr/local/apcupsd/apcupsd.css
/usr/share/transmission/web/style/jqueryui/jqueryui-1.8.16.css
/usr/share/transmission/web/style/transmission/common.css
/usr/share/transmission/web/style/transmission/mobile.css
/www/css/reset.css
/www/css/style.css
-----------------------------------------------------------------------------------------------------------------------------

Replication steps

Setup Wireless
Enabled Captivate Portal on Wireless Lan Segment
Enable Captivate Site forwarding. 
Connect to Wireless on mobile/computer device
Try and Sign into Wireless. Browser will load to blank white page

Jackysi, 2015-03-05:
Apologies for such long delay answering your question. I been busy =/. I will give it a look.
Thank you for detailed report, this will help me find the issue quickly =)

Jackysi, 2015-03-08:
Issue fixed for the next GUI version 2.6 which will be released with new Shibby update (waiting for it)

closed 2015-03-08 by Jackysi