Jackysi/advancedtomato #39: QoS graphs missing R7000

tonyarnold, 2015-03-09:
When accessing **Quality of Service &rarr; View Graphs** the following error is shown:

```
showData
init
eval code
eval@[native code]
globalEval@http://192.168.2.1/js/jquery.min.js:2:4266
domManip@http://192.168.2.1/js/jquery.min.js:3:21401
append@http://192.168.2.1/js/jquery.min.js:3:18993
http://192.168.2.1/js/jquery.min.js:3:20363
access@http://192.168.2.1/js/jquery.min.js:2:5806
html@http://192.168.2.1/js/jquery.min.js:3:20026
onCompleted@http://192.168.2.1/js/advancedtomato.js:224:40
onReadyStateChange@http://192.168.2.1/tomato.js:1603:17
http://192.168.2.1/tomato.js:1855:38Can't find variable: qrates_out
```

Jackysi, 2015-03-10:
I commented on this issue on Linksysinfo forums. I forgot to include ARM qos graphs instead I added AC ones. Its just invalid file compile :) I fixed this with new version and also already updated GIT with fix.

closed 2015-03-10 by Jackysi

tonyarnold, 2015-03-10:
Excellent! Thanks @Jackysi! I look forward to the update.