Jackysi/advancedtomato #40: QoS is disabled no matter what I do

homer2320776, 2015-03-15:
Upgraded to AT 2.6-127 from standard Shibby 127ARM. My router is an AC68U and QoS seemed to work okay on shibby mod, but on AT, the transfer rates under QoS is blank along with the QoS details page.

Clicking on B/W Limiter it shows QoS is disabled at the top of the screen.

Jackysi, 2015-03-21:
Hmm, strange! I will give it a look as soon as I have time.

Jackysi, 2015-07-05:
I tested your issue through and couldn't re-create it. I would guess you did not reset NVRAM values after upgrading from Shibby. I recommend doing that or upgrading to 2.8-130 which may fix the issue (this is the version of firmware where I tested the issue). Please reopen if issue did not go away!

closed 2015-07-05 by Jackysi