Jackysi/advancedtomato #41: Can't save IPv6 form

bjeanes, 2015-03-20:
Asus RT-AC68U with Tomato Firmware 1.28.0000 -2.6-127 K26ARM USB AIO-64K installed yesterday.

When trying to save the IPv6 form with either Native IPv6 or DHCPv6 w/ prefix delegation, nothing happens. The browser inspector shows an error because `E('_fom').ipv6_isp_opt` is `undefined`. If I manually comment out that line, it blows up later because `E('_fom').ipv6_vlan` is also `undefined`.

Let me know if there's more info I can give.

Jackysi, 2015-03-21:
It is possible I forgot to include some form fields when updating page, I was in rush and I apologize. I will give it a look when I have free time.

Jackysi, 2015-04-13:
I am sorry for late response, this issue is supposed to be fixed in 2.6-128. Can you confirm?

bjeanes, 2015-04-13:
I upgraded to Tomato Firmware 1.28.0000 -2.6-128 K26ARM USB AIO-64K. IPv6 form still won't save. Same error... `fom.ipv6_isp_opt.value = fom.f_ipv6_isp_opt.checked ? 1 : 0;` throws an exception because `fom.ipv6_isp_opt` is undefined. This is on the latest Chrome on OS X

bjeanes, 2015-04-14:
Out of curiosity I flashed the NVRAM and used a different browser, in case the .js was being cached in some way. Turns out, config restore does not work. The config backup I get is not a valid gzip file, as my googling would suggest it should be. Is this a known issue? 

In any case, this page is definitely still not working.

Jackysi, 2015-04-14:
I did heard of the issue with config file getting corrupted. But I honestly have no idea how and why. It does not seem to be GUI related. It makes no sense that it could be anyway.

I fixed IPv6 issue, see commit: https://github.com/Jackysi/AdvancedTomato-V2/commit/4cd7a9119194774b5fd7ff669c103fe6bdc6a91f

closed 2015-04-14 by Jackysi

bjeanes, 2015-04-14:
@Jackysi np, I'll manually verify backup files and do my own `nvram` dumps for now. Is there a good place to report that issue. Maybe I can assist in debugging it as it is 100% reproducible, even on latest firmware.

As for IPv6, I enabled it directly in nvram. I'm getting IPs now but can't ping most public IPv6 addresses. I think this is separate from the form issue. I'll check the form again when a new firmware is released. Thanks.