Jackysi/advancedtomato #42: Saving certain settings redirects to port 80

nemchik, 2015-03-20:
I have my router's web interface remotely accessible and running on a non standard port. In the admin pages, some settings when saved display a white screen with "Working, please wait..." and a count down. Once these pages complete their count down i am redirected to my router but on port 80. this results in 404 as my router is not open on port 80.
I do not know if this was an issue on the original shibby mod but I was using his 124 build until today i switched to AT and i do not remember experiencing this issue on shibby's builds.

i am using an asus rt-ac66u on your 2.5-124 build btw

Jackysi, 2015-03-21:
Thanks for report, I will give it a look as soon as I am free.

Jackysi, 2015-04-13:
Strange, I was not able to re-produce this issue. I tried on both routers Asus RT-N66U and R7000. Both worked fine redirecting to port 8888 (non standard http port)

nemchik, 2015-04-13:
I can confirm this issue does exist by navigating to https://my.public.IP.or.domain:8080/admin-access.asp and clicking save (not making any changes). It does this every time. I've also confirmed the issue exists in shibby original 128, not just advanced tomato.

Jackysi, 2015-11-02:
Apologies for very very late response, I will give it a look soon I am just having issues with compilation and everything!

Jackysi, 2015-12-07:
I am unable to replicate this issue in version 3.0-132. Can you verify please?

nemchik, 2015-12-07:
I'm back on Shibby's (non advanced tomato) for now and on 132 I can confirm the issue still exists. Have you done something different/specific to fix this in your mod?

nemchik, 2015-12-07:
I should also mention (I didn't think it was relevant but it might be) I have external access set to HTTPS only (no HTTP) so that might be why. It's redirecting from https://my.public.IP.or.domain:8080/admin-access.asp to  https://my.public.IP.or.domain/admin-access.asp (it completely omits the port).

Jackysi, 2016-12-03:
Weird is I'm using port 444 and I have no issue like that. If you could point to exact page you experience issue with I might figure it out, otherwise this issue will stay open indefinitely.

nemchik, 2016-12-03:
I have temporarily switched to Merlin's firmware (don't worry, I'll be back). I remember it happening on the administration page where you could set the admin password or configure ssh access and telnet. It's been a while since I've bothered with it, may have been fixed by shibby by now. I won't have time to test for quite some time (I'm moving this month) so it's up to you if you feel this issue should be closed.

Jackysi, 2016-12-03:
I will give it a test :) but its likely that this redirect is done via HTTPD handler where my knowledge is limited.

Jackysi, 2017-05-08:
I no longer have this issue with latest firmware, can you verify please?

nemchik, 2017-05-08:
I'll give it a shot on a spare I have but it's a lower end model so I'll have to test it with shibby's build. I think if it's solved there it'll likely be solved here as well.

nemchik, 2017-05-16:
Ok. I've confirmed this issue still exists, but upon reading everything I've posted here I realize I've posted some incorrect and poorly described issues.

So here's my attempt to rectify:

I navigate to https://fakesubdomain.nemchik.com:8443/#admin-access.asp from the outside (Google chrome on my cell phone with WiFi disabled)
My settings are as follows:
Local access: http & https
Http port: 80
Https port: 443
Remote access: https
Port: 8443
(Other irrelevant settings)
I only certain settings on this page are important enough to cause the issue when changed. For example, I have remote SSH enabled, and my remote port for SSH is 22, but if I change the remote SSH port to 2200 (or anything, just making a change) and then scroll down and hit save, I am redirected to https://fakesubdomain.nemchik.com/#admin-access.asp which is actually port 443, which is the standard https port.

I had previously stated I was being redirected to port 80, which is incorrect.

Would there be any kind of logs I could grab that might help?

Jackysi, 2017-05-16:
Very strange. I use port 444 on my Tomato and redirect works fine on Save. I'm curious what could cause that. Logs probably don't say anything about that cause http doesn't log these things. I will research this further by looking source and gonna let you know what I find out!

Jackysi, 2017-05-16:
Did you try clearing NVRAM? That could be the case?

nemchik, 2017-05-16:
Yeah I've cleared nvram and tested only setting the needed things to test this and otherwise everything is default out of the box.

When I first opened this issue I had my remote admin port set to 8080 but have since switched to 8443, so I don't think the port I choose makes a difference.

The page with the countdown seems to be tomato.cgi but I'm also not well informed about how this works, so that might not be the right place to look.

tomdotorg, 2017-08-15:
Seeing the same thing. Running "Tomato Firmware 1.28.0000 MIPSR2-3.4-140 K26 USB AIO"

I run the external admin on port https/8080. When I change the theme color (Admin Access --> User Interface Settings --> Interface Theme) and save it, it redirects to the right URL but missing the port number, so it doesn't connect back.

Happy to provide more info if it would help.

Jackysi, 2017-08-15:
Its a bug in Tomato HTTPD software which is doing the redirect. I am aware of it (I already seen it as well). I will try to fix it once I have time.