Jackysi/advancedtomato #44: Unable to create virtual wireless (w0.1)

brentster101, 2015-03-26:
Netgear R7000
v2.6-127

Able to create virtual wireless interface w0.1 but unable to save.  When clicking save and returning to overview page, w0.1 is not present there or on virtual page,

Jackysi, 2015-04-06:
Have u tried new 2.6-128 update?

CyberSho, 2015-04-07:
Bug still present in 2.6-128 on RT-N18U and Win7/IE11. For me eth1 WLAN is in client mode and I want to add second virtual WLAN in AP mode. Cannot save this configuration, but it was posssible and working on shibby 128 build, so it's the AT GUI issue only. It's another case to allow changing modes on virtual WLAN to other than AP, but I have to write to shibby for that.

Jackysi, 2015-04-08:
This is helpful, I will look into it. It might take few days got a project on hands but as soon as possible I will fix it. Thanks!

CyberSho, 2015-04-19:
I'm sorry for delay.

Error is on code:
if (elem[i].name.indexOf('wl' + u) == 0) {
s += 'nvram set ' + elem[i].name + '=\'' + elem[i].value + '\'\n';
}

Exception text is:
Unable to get property 'indexOf' of undefined or null reference

Jackysi, 2015-04-21:
That is very helpful, I will give it a look when I have time. I hope I will be able to fix this for next release.

atx32, 2015-06-23:
Just a note that I am having the same issue on an Asus RT-N66U and v2.8-130 (tomato-K26USB-1.28.AT-RT-N5x-MIPSR2-2.8-130-AIO-64K). I noticed that the problem occurs with IE11 but not with Chrome. The plain Shibby build works fine with IE11.

I hope you can get this fixed in a future release.

Jackysi, 2015-07-08:
I am unable to re-create the issue running IE12 (EDGE windows 10 thing). Sorry but I don't really care about IE11 so I won't bother fixing this issue.

craiggoodwin, 2016-06-17:
Hi,

This might be an old thread but I am having the same issue.  
Hardware.  Netgear R7000
firmware: Tomato Firmware 1.28.0000 -3.2-136 K26ARM initial-64K

I can set up the vlan and bridge fine but as soon as I try to add a virtual wireless I get problems, where I click save and although it says saving it just hangs, never appears to complete the save and the network isn't there when I go back to check.

I've cleared the NVRAM as well as rebooting the router with no luck.  somewhat frustrated as this was one of the reasons I decided to use tomato advanced.

Thanks

Jackysi, 2016-10-02:
I was able to create virtual interface on my R7000 on 3.3-138. Can someone verify?

AnderssonPeter, 2017-01-03:
I just created a guest wifi network on my R7000 running 3.4-138 and it worked using Edge (im not at home right now so i can't tell you the exact version).

closed 2017-11-14 by Jackysi