Jackysi/advancedtomato #45: ARM-only features are incorrectly "present" on mipsel builds

AndreDVJ, 2015-04-05:
Hey Jacky,

There's some ARM-only stuff like ext4 filesystem and USB 3.0 support on the GUI. Please check the sources. I have a mipsel router, and messing around the source code I found them.

Jackysi, 2015-04-06:
I don't know where these features are enabled/disabled etc etc... I only merged code and compiled. So this issue must have came from Shibby's GIT. What I am trying to say is I don't have the knowledge to fix this. I can only wait for Shibby's GIT updates.

AndreDVJ, 2015-04-06:
Yeah came straight from Shibby's ARM GIT. I have redone my whole "environment" to use Debian 7.8 and cloned both your repo and shibby's mipsel one.

I spent yesterday night messing around the code and fixing things for myself (and recompiling images). I am attaching two pastebin links with about.asp (which I updated to conform with shibby's current sources) and nas-usb.asp with removal of ARM stuff.

http://pastebin.com/272Y4G1j

http://pastebin.com/aDp5W4Jz

Jackysi, 2015-04-08:
Oh, The GUI only. Sorry I didn't answer sooner I am just extreme busy!
If you mentioned before that GUI shows options which are not there I'd understand it better :P I was sure you were talking about internal options not GUI it self. I will take a look and strip it out :D

Jackysi, 2015-04-14:
Fixed in commit: https://github.com/Jackysi/AdvancedTomato-V2/commit/486e00d5cd81f4a26516e0c52578612437f2cd0e
Thanks for report :)

closed 2015-04-14 by Jackysi