Jackysi/advancedtomato #46: [2.7-128] QoS Graphs throwing an error

tonyarnold, 2015-04-15:
When I try to access the QoS graphs page, I see the following error on my R7000:

```
showData
init
eval code
eval@[native code]
globalEval@http://router.local./js/jquery.min.js:2:4266
domManip@http://router.local./js/jquery.min.js:3:21401
append@http://router.local./js/jquery.min.js:3:18993
http://router.local./js/jquery.min.js:3:20363
access@http://router.local./js/jquery.min.js:2:5806
html@http://router.local./js/jquery.min.js:3:20026
onCompleted@http://router.local./js/advancedtomato.js:225:40
onReadyStateChange@http://router.local./tomato.js:1603:17
http://router.local./tomato.js:1855:38Can't find variable: qrates_out
```

Jackysi, 2015-04-16:
For some reason the invalid QOS Graphs file was compiled with ARM images. It should be ARM specific but for some reason I added the RT-AC one. I will re-compile ARM images today and re-upload under same name, only different MD5.

Jackysi, 2015-04-16:
Images have been recompiled, if you use QOS-Graphs page please re-flash router with latest image from the web site. The name is the same, but MD5 hash changed.

closed 2015-04-16 by Jackysi

tonyarnold, 2015-04-16:
Thanks, @Jackysi — worked a treat, I have graphs back :)