Jackysi/advancedtomato #49: Web Services Web Server page error when starting/stopping server

jbobthegreat, 2015-05-03:
Starting or stopping the nginx webserver results in an error page every time.  The server itself does successfully start and behaves normally, but I have to click on the web services link in the sidebar menu again to display the configuration settings.  This occurs every time I start or stop the server.  

[image]

Jackysi, 2015-05-03:
Thanks for report, can you include your AdvancedTomato version please?

jbobthegreat, 2015-05-03:
I'm running 2.7-128 K26ARM USB AIO-64k on a Netgear R6250

Jackysi, 2015-07-05:
Apologize for late response, fixed issue for next release.

closed 2015-07-05 by Jackysi