Jackysi/advancedtomato #50: Firmware update broken.

mattaltus, 2015-05-04:
Hey,  I have a Netgear R7000 that was running AT 2.7-128.  I was getting a few errors as described in #49, also transmission wouldn't stop/start via your interface as well, but that's for another day.  Anyway back to the problem, I decided to move back to shibby's build.  However flashing via the web interface didn't work. I selected the firmware image hit upgrade, clicked the confirmation box, then got a 500 error.  The router didn't reboot and seemed to be working ok (wifi, ppp. ssh were all still working), but the load spiked and downloads became very slow.  A manual reboot didn't help either.

I managed to do the upgrade by scp'ing the image and running 
```
mtd-write2 tomato-R7000-ARM--129-AIO-64K.trx linux
```
This got me back to a stable state running shibby's 129.

So I looked into the problem and it seems that your upgrade code wants to use 'mtd-write' to flash the router while shibby uses mtd-write2.  And your latest image doesn't have mtd-write installed only mtd-write2..

This [upgrade.c](https://bitbucket.org/pl_shibby/tomato-arm/src/eeadb0194733b0fd3fd9db2b93de365461649d6a/release/src-rt-6.x.4708/router/httpd/upgrade.c?at=shibby-arm) is the latest from shibby and uses mtd-write2 while your [upgrade.c](https://github.com/Jackysi/advancedtomato2/blob/advancedtomato-ac/release/src/router/httpd/upgrade.c) uses mtd-write.

May just be related to ARM builds not MIPS, haven't looked into it anymore.  Anyway hope this helps.

Jackysi, 2015-05-04:
Am, that's strange. It works just fine for me on R7000... I never touched these files... 
But though you are comparing incorrect gits, the Github repo is for RT-AC / RT-N. ARM is here: https://bitbucket.org/jackysi/advancedtomato-arm/src/2e20967ca5d4a5cce09b7d5dba59dc732c064db3/release/src-rt-6.x.4708/router/httpd/upgrade.c?at=master

And it uses the same method as Shibby.

mattaltus, 2015-05-04:
Oops wrong spot. Maybe the problems were similar to the other general http type errors I was seeing then. In that case it may be a dupe of that other issue.

Jackysi, 2015-05-04:
Its possible yes. I did have same problem once. Only once. The fix was to re-try upgrade and it worked the second time. Any how, its not AdvancedTomato issue so I can close it :). I know this doesn't solve anything, but even if I wanted to I can't fix this issue. My Knowledge is too limited sorry :(

closed 2015-05-04 by Jackysi

mattaltus, 2015-05-06:
FYI: I figured out the problem.  There was a problem with ipv6 dhcp which caused dhcp6c to constantly restart dnsmasq and httpd.  So as I started the upgrade, httpd was killed which cancelled straight away, I was pretty lucky I wasn't left with a corrupt image on the flash.  This would have caused all the other http type errors I was seeing in the web interface too.

Easy workaround is to disable ipv6 and then do the upgrade.

Longer term work around is listed [here](http://www.linksysinfo.org/index.php?threads/ipv6-and-comcast.38006/#post-260959). Hopefully shibby can do an update sometime soon, in the mean time my script solution keeps things working.