Jackysi/advancedtomato #52: "Invalid IP address" for valid IP in port forwarding config

austinholbrook, 2015-05-19:
I use a random /22 for my LAN in the 10.x range of IP addresses, primarily to avoid IP address conflicts when connecting to VPNs.

The router is on 10.213.200.1, and the subnet mask is 255.255.252.0. The network address is 10.213.200.0 and the broadcast address is 10.213.203.255. All addresses in-between those two are routable, I have a server sitting on 10.213.201.0 and it works just fine.

The GUI doesn't approve of that however, and trying forward some ports to the server's internal IP results in an "Invalid IP address" error. It looks like it happens anytime you use an address that ends in either a 0 or 255, when it should only be considering the network and the broadcast address as invalid.

Ivantosov, 2017-05-08:
Is there any news?

Jackysi, 2017-11-14:
I don't know how to fix this because I don't know the formulas they use to parse IP Addresses. So I will leave this one to someone that does know.