Jackysi/advancedtomato #55: OpenVPN client stop/start lands on vpnclient.asp

tparvu, 2015-07-03:
I recently updated to 1.28.0000 -2.8-130 K26ARM USB AIO-64K on an Asus RT-AC68R/U.

When I start or stop an OpenVPN client instead of reporting the status I land on http://10.10.10.1/vpn-client.asp which just displays 2 buttons, "save" and "cancel".

[image]

The buttons don't appear to do anything. The OpenVPN client will start/stop as can be seen by going to the status URL. It doesn't matter if you hit the buttons or do nothing at all. 

This happens every time I start or stop an OpenVPN client. 

Thanks for the great software. The stock Asus firmware was unstable and frustrating. Router is working great with AdvancedTomato!

Jackysi, 2015-07-03:
I will fix this in next release, thanks for report :)

tparvu, 2015-07-03:
Thanks much!

Jackysi, 2015-07-03:
No problem, thank you! fixed :)

closed 2015-07-03 by Jackysi