Jackysi/advancedtomato #56: OpenVPN client fails to configure, Invalid IP and Port errors

SenorLAN, 2015-07-16:
When I try to save the configuration for the OpenVPN client (client 1 or 2), I receive two errors: "Invalid IP Address" and "Invalid Number. Valid Range: -1-32767."

[image]
[image]

This was the first time to configure the OpenVPN clients on this router. The initial config seems to save fine, but then any further changes to the setup (i.e. protocol, port, or custom configuration)  give the two errors and do not save.  

My VPN provider is AirVPN and authentication is done with certificates. Below are the configuration pages as configured. 
[image]
[image]
[image]

Troubleshooting: 
If I change the Authorization Mode to Static Key and fill in any IP addresses into the Local/Remote endpoint fields then I no longer get the IP address error, but I still get the port error.
If I change the Server Address/Port port# it has no effect on the errors. 
I tried using VPN Client 2 to set up the same configuration and I receive the same errors with that client as well. 

I am using MIPSR2-2.8-130 K26 USB AIO on an ASUS RT-N16. This router was stock ASUS firmware before loading this AdvancedTomato build. All instructions to clear the NVRAM and any stored configs were followed as part of the flashing/conversion process. 

The router works fine in all other regards that I am aware of. The configuration was validated against the VPN provider guide for setup on Tomato. The VPN provider works fine if I use a desktop client or my PFSense server to connect to it.

NYPD, 2015-08-18:
I could perhaps be running into sort of the same issue, but my problem is the inability to start the client.  I'm using the latest 2.9-131 on the R7000. I still need to test more and actually see if i can narrow something down.

vladoh, 2016-02-20:
similar issue here. When I have configured just Server nr.1 as TUN everything works fine. I wanted to start also Server nr.2 as TAP. After pasting keys it seems nvram got overwritten somewhere. I lost port setting in Server nr.1 and was not able to recover. Even after correcting port in Server nr.1 and starting it, it reverted back to empty value. I have received also same weird messages regarding missing port or incorrect value in TLS Renegotiation Time which was removed also. I had to restore old configuration to at least have Server nr1. running.

Knifa, 2016-11-04:
Same issue here. Tomato Firmware 1.28.0000 MIPSR2-3.3-138 K26 USB AIO.

skywalker-greg, 2016-12-07:
In my case, the second error "Invalid number ..." appears to be not for a port number, but for "Connection retry" entry.

nfonseca, 2016-12-12:
Same issue running on Tenda N6 Tomato Firmware 1.28.0000 MIPSR2-3.4-138 K26 USB TendaN6
Is there any fix workaround ?

nfonseca, 2016-12-12:
I have noticed that nvram is full when I get that error

[image]

nfonseca, 2016-12-13:
I found out out to workaround this bug. if you free up space on NVRAM by deleting some PortForwarding rules that are set up as examples, you won't have that problem anymore.
I think it's because the nvram gets full.

skywalker-greg, 2016-12-14:
>  if you free up space on NVRAM by deleting some PortForwarding rules that are set up as examples, you won't have that problem anymore.

Thanks, man! It works!
Tomato Firmware 1.28.0000 MIPSR1-138 K26 USB VPN

nfonseca, 2016-12-14:
Nice ! I think the NVRAM is a bit short on the Tenda N6 but deleting some unused stuff can help !

Jackysi, 2016-12-14:
So if I understand properly, this is not actual issue =). NVRAM is actually "database" for all router options. Every command, script (admin scripts), port forward, etc... Is stored there. So 32kb isn't really big if you have long paths, lots of scripts, many additional commands and so on. Sadly some routers don't have enough storage available to have more NVRAM (Tenda N6 in this case). I stopped supporting 8mb flash size routers so honestly Asus RT-N16 and Tenda N6 are huge exceptions. 

I don't have much special stuff on my R7000 and my NVRAM usage is: 41.84 KB / 64.00 KB (65.37%)

nfonseca, 2016-12-14:
As long we can bypass the issue freeing up some space it's OK. Anyway these devices are Home Router so they should not have a lot of Customizations. In my case I'm runing 91% nvram which makes it work as I want. Just don't stop supporting these devices. The Tomato Software is just great !

nfonseca, 2016-12-14:
You can mark this as "not a bug" yes. It's a limitation of the NVRAM

vladoh, 2016-12-14:
NVRAM might be really an issue. I will try it later but even after removing examples from port forwarding I'm getting 93.25% NVRAM :(. Would be nice to have some feature in GUI as "NVRAM detailed usage" to see what exactly is consuming the space in NVRAM.

nfonseca, 2016-12-14:
There are many other things you can clear to make the NVRAM thinner. Not only Port Forwarding. Look at QOS Rules :-)

Jackysi, 2016-12-14:
Detailed NVRAM usage, not sure how would you show that, there is many different values (few hundred) only very few are actually large. In NVRAM is very simple database its almost as easy as "STRING = VALUE". You can export NVRAM and you will see which "value" is longest. Administration -> Debugging -> Dump NVRAM

closed 2017-11-14 by Jackysi