Jackysi/advancedtomato #63: advanced-routing.asp broke in chrome browser

cubieb, 2015-08-27:
advanced settings> routing> 
 it will show error like below this, routing table can't be shown.
TypeError: Cannot set property 'disabled' of null
    at verifyFields (eval at  (http://192.168.1.1/js/jquery.min.js:2:4243), :100:25)
    at eval (eval at  (http://192.168.1.1/js/jquery.min.js:2:4243), :1:14)
    at eval (native)
    at Function.x.extend.globalEval (http://192.168.1.1/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://192.168.1.1/js/jquery.min.js:3:21391)
    at x.fn.extend.append (http://192.168.1.1/js/jquery.min.js:3:18985)
    at null. (http://192.168.1.1/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://192.168.1.1/js/jquery.min.js:2:5780)
    at x.fn.extend.html (http://192.168.1.1/js/jquery.min.js:3:20020)
    at Object.TomatoAJAX.onCompleted (http://192.168.1.1/js/advancedtomato.js:233:29)Cannot set property 'disabled' of null

somebody found this ?

Jackysi, 2015-08-27:
Could you please share your AdvancedTomato version?

cubieb, 2015-08-27:
function verifyFields(focused, quiet)
		{
			/* VLAN-BEGIN */
			if (E('_f_dr_lan') != null) {
				E('_f_dr_lan').disabled = (nvram.lan_ifname.length < 1);
				if (E('_f_dr_lan').disabled)
					E('_f_dr_lan').checked = false;
			};
			if (E('_f_dr_lan1') != null) {
				E('_f_dr_lan1').disabled = (nvram.lan1_ifname.length < 1);
				if (E('_f_dr_lan1').disabled)
					E('_f_dr_lan1').checked = false;
			};
			if (E('_f_dr_lan2') != null) {
				E('_f_dr_lan2').disabled = (nvram.lan2_ifname.length < 1);
				if (E('_f_dr_lan2').disabled)
					E('_f_dr_lan2').checked = false;
			};
			if (E('_f_dr_lan3') != null) {
				E('_f_dr_lan3').disabled = (nvram.lan3_ifname.length < 1);
				if (E('_f_dr_lan3').disabled)
					E('_f_dr_lan3').checked = false;
			};
			if (E('_f_dr_wan') != null) {
				E('_f_dr_wan').disabled = (nvram.wan_proto == 'disabled');
				if (E('_f_dr_wan').disabled)
					E('_f_dr_wan').checked = false;
			}
			/* VLAN-END */
			return 1;
		}
i modified like that , that error disappear.

cubieb, 2015-08-27:
I cloned from github yesterday, used the newest —ac branch,build tenda n6 firmware, accturely I did not change code before compile.

eternity7602, 2015-09-11:
I apologize for my unfamiliarity, but how do I make the recommended change to repair the error? This is the error I receive when I attempt to access the routing section under advanced. I am using advanced tomato version "2.8-130 K26 USB WNDR64" on my Netgear WNDR3400v1.

"There has been error while loading a page, please review debug data bellow if this is isolated issue.
If this is not an isolated issue, please create issue with details bellow and how to reproduce the error at	https://github.com/Jackysi/advancedtomato2/issues. 

Detailed information:
TypeError: Cannot set property 'disabled' of null
    at verifyFields (eval at  (http://192.168.1.2/js/jquery.min.js:2:4243), :100:25)
    at eval (eval at  (http://192.168.1.2/js/jquery.min.js:2:4243), :1:14)
    at eval (native)
    at Function.x.extend.globalEval (http://192.168.1.2/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://192.168.1.2/js/jquery.min.js:3:21416)
    at x.fn.extend.append (http://192.168.1.2/js/jquery.min.js:3:18985)
    at null. (http://192.168.1.2/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://192.168.1.2/js/jquery.min.js:2:5780)
    at x.fn.extend.html (http://192.168.1.2/js/jquery.min.js:3:20020)
    at Object.TomatoAJAX.onCompleted (http://192.168.1.2/js/advancedtomato.js:225:36)Cannot set property 'disabled' of null

Refreshing browser window might help."

I have tried to access this section using Chrome, Edge, Firefox, and Dragon browsers with no luck. Until I swapped in my R6300v2 this was not a concern, but now I intend to use the WNDR3400v1 to extend my network. Thank you in advance for any help.

Jackysi, 2015-09-11:
I don't have a router to test out these issues, it works fine on R7000 and RT-N66U. I will have to compare code with Shibby's to see what I missed out. You can't fix the problem on your own if you don't have the specific knowledge

eternity7602, 2015-09-14:
ok, thank you for the quick response and any efforts you are able to put into it. It is much appreciated.

let me know of anything i can do to help out.

eternity7602, 2015-09-17:
just a small update... i found the update to version 131 and applied it, so i am now using a WNDR3400v1 running tomato-K26USB-NVRAM64K-1.28.ATMIPSR2-2.9-131-WNDR64-WNDR3400 and the issue remains when trying to open the advanced settings/routing section. 

I am attempting to use this router to extend the coverage of my R6300v2. My understanding is that there are things in that section that need to be changed. Thanks for any progress made so far and for the replies.

Jackysi, 2015-09-17:
It may be connected to your device specifically which is a problem for me because I can't debug the issue. AT is postponed at the moment because I have busy schedule but when I will get back to it I will check the problem and try to find the issue.

closed 2016-10-02 by Jackysi