Jackysi/advancedtomato #65: SSH not working on DIR868L-AT-ARM-2.9-131-special.trx

dlaporte, 2015-08-29:
I installed Advanced Tomato on my DIR-868L and am loving it!  I have one minor issue - the SSH service is not working.  SSH is set to enable at startup, but it shows as (stopped).  Manually attempting to start it with the play button doesn't change the state.  Logging in and poking around via telnet, it looks to me like the dropbear binary is missing from the build.  Is that by design?

thanks!

Jackysi, 2015-08-30:
That sounds rather strange. It is possible that Shibby did some changes there because I can use SSH just fine on Asus RT-N66U and Netgear R7000. DIR-868L has some space issues so its possible that Shibby cut corners there. But you could try to flash older firmware version or just re-flash most current and see if that changes.

Jackysi, 2015-09-11:
Try writing a topic at http://www.linksysinfo.org/index.php?forums/tomato-firmware.33/

closed 2015-09-11 by Jackysi