Jackysi/advancedtomato #68: Random errors when loading Overview

evakq8r, 2015-09-12:
Since the 2.9.131 update, I've been getting random errors such as:

TypeError: Cannot read property 'length' of undefined
    at eval (eval at  (http://192.168.1.254/js/jquery.min.js:2:4243), :6:33)
    at eval (native)
    at Function.x.extend.globalEval (http://192.168.1.254/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://192.168.1.254/js/jquery.min.js:3:21391)
    at x.fn.extend.append (http://192.168.1.254/js/jquery.min.js:3:18985)
    at null. (http://192.168.1.254/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://192.168.1.254/js/jquery.min.js:2:5780)
    at x.fn.extend.html (http://192.168.1.254/js/jquery.min.js:3:20020)
    at Object.TomatoAJAX.onCompleted (http://192.168.1.254/js/advancedtomato.js:231:29)
    at Object.XmlHttp.onReadyStateChange (http://192.168.1.254/tomato.js:1603:6)Cannot read property 'length' of undefined

Spam refreshing the window sometimes fixes it, but otherwise it recurs pretty frequently. Page quickly  becomes unresponsive and is replicable across multiple browsers (IE, Firefox, Chrome etc).

evakq8r, 2015-09-12:
Additionally:

Jackysi, 2015-09-17:
Rather strange, I never experienced this problem before. Did you try to flash image and clearing NVRAM? I am pretty sure there is something from NVRAM causing this.

evakq8r, 2015-09-17:
Yep, each flash I do of a new Firmware I factory reset the router including NVRAM. The issue has been intermittently recurring since 2.9.129 but now occurs every time I try and load the gateway.

evakq8r, 2015-09-17:
Am happy to try it again and flash from scratch if need be, but won't be able to until I get home from work.

evakq8r, 2015-09-23:
Welp, as I hadn't heard anything decided to do the F/R and reflash anyway.

- No config other than ISP settings and WiFi setup, still get the same errors:

ERROR occured!
There has been error while loading a page, please review debug data bellow if this is isolated issue.
If this is not an isolated issue, please create issue with details bellow and how to reproduce the error at	https://github.com/Jackysi/advancedtomato2/issues. 

Detailed information:
TypeError: Cannot read property 'length' of undefined
    at eval (eval at  (http://192.168.1.254/js/jquery.min.js:2:4243), :6:33)
    at eval (native)
    at Function.x.extend.globalEval (http://192.168.1.254/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://192.168.1.254/js/jquery.min.js:3:21391)
    at x.fn.extend.append (http://192.168.1.254/js/jquery.min.js:3:18985)
    at null. (http://192.168.1.254/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://192.168.1.254/js/jquery.min.js:2:5780)
    at x.fn.extend.html (http://192.168.1.254/js/jquery.min.js:3:20020)
    at Object.TomatoAJAX.onCompleted (http://192.168.1.254/js/advancedtomato.js:231:29)
    at Object.XmlHttp.onReadyStateChange (http://192.168.1.254/tomato.js:1603:6)Cannot read property 'length' of undefined

Refreshing browser window might help.

evakq8r, 2015-09-23:
That's replicable off Firefox, Chrome and IE/Microsoft Edge.

Jackysi, 2015-09-23:
I don't have AC66U to test, but on N66U I can't replicate these issues, nor on R7000. Its very hard to debug something you don't have :(

evakq8r, 2015-09-23:
It's an AC68U*, but I get the idea.

evakq8r, 2015-09-23:
Oh well, guess I'll put up with it for now. :(

Jackysi, 2015-09-23:
Ow ye sorry. It may be something device specific like Tomato issue (firmware httpd handler is messing with you), i don't know. The error itself only says that some element is missing. Tomato original GUI is very terrible at handling errors and my work is based on that GUI. I have fixed so many issues but re-coding the GUI is just too much work. So i'm fixing broken stuff as I go. Sometimes somethings are extremely hard to fix/debug without the device. I will have in mind these issues and report, I won't forget. In next update I will change the way pages are loaded (the core AJAX function) so hopefully it will be more "soft" on these sort of errors

evakq8r, 2015-09-23:
No worries, thanks Jacky. Appreciate you responding either way.

Jackysi, 2015-09-23:
np, thank you for report.

totalretribution, 2015-09-26:
I'm getting this same issue on my Linksys E4200v1 using Firefox, chrome and edge. It also however often does not even load the AdvancedTomato webpage.

The strange thing is that it does not happen on my android tablet.

evakq8r, 2015-09-26:
I did a bit more tinkering on my issues, and it seems the errors are less frequent when IPv6 is completely turned off (they are still there, but it's not as common).

Jackysi, 2015-12-07:
I can not replicate this issue no matter how I try. This must be hardware/Firmware related. Did you attempt to check console after 3.0-132 for more details on the problem (make sure you click the actual message to see "invalid" code instead of just reading log which makes zero sense).

closed 2016-01-08 by Jackysi

evakq8r, 2016-01-12:
Hi jackysi, 

Sorry for my lack of contact, forgot about this issue.

I've ended up going with another firmware as a result of these errors, haven't tried since.

If I end up testing again and it still populates the error, I'll let you know with another error submission.