Jackysi/advancedtomato #69: PPTP Inbound client connection dropped when logging inbound traffic

SimplyAutomationized, 2015-09-14:
When enabling connection logging for inbound traffic any new clients to the pptp vpn will be dropped. (current connected clients will stay connected until they reconnect. resulting in a drop).
Even after the pptp server is restarted the clients can't connect unless I disable inbound connection logging for drops.

[image]

Tomato Firmware 1.28.0000 MIPSR2-2.9-131 K26 USB Mega-VPN
Model	Linksys E4200 v1
Chipset	Broadcom BCM4716 chip rev 1 pkg 10
CPU Freq	480MHz
Flash Size	16MB
Details of IPTABLES, netstat, and processes:

[image]
[image]
[image]
[image]

Jackysi, 2015-09-17:
I am sorry to inform you this: The issue is not related to AdvancedTomato Interface but Tomato itself. Please address the issue to Shibby since he is the main developer of Tomato mods I only fork his work and create GUI for it.

closed 2015-09-17 by Jackysi

SimplyAutomationized, 2015-09-17:
OK thanks :-)