Jackysi/advancedtomato #71: D-Link DIR-868L won't load .trx file

baltojeff, 2015-09-16:
I've downloaded the latest version, tomato-DIR868L-AT-ARM-2.9-131-special.trx, file and confirmed the checksum. When trying to upload the file to my router, I get the following error: 
[image]
I've tried changing the file type to .bin and get the same error. Any thoughts as to what I should be doing different? Hardware v.A1 Firmware v1.09

Jackysi, 2015-09-17:
I have no idea what could that be... Can you flash Shibby's images?

aviadsp, 2015-09-25:
HI, I also have Dir-868L rev.a
I was install Advanced-tomato successfully in emergency login of the router.
file (tomato-DIR868L-AT-ARM-2.9-131-special.trx).

Jackysi, 2015-10-02:
As far as I know DIR868L should be flashed via Emergency Login and not standard router GUI. So this is not an actual issue just incorrect way of flashing. Also I don't offer support for Tomato Firmware issues only Graphical Interface issues which are part of AdvancedTomato GUI.

closed 2015-10-02 by Jackysi