Jackysi/advancedtomato #73: Direct links to Bandwidth page (with username and password) no longer working

arigit, 2015-09-21:
Up until release 2.7-128, the following link would work perfectly in any browser:

http://my_username:my_password@192.168.1.1/#bwm-24.asp

It would login and display the graph without prompting for a username/password.

I used this in a shortcut in my HTPC, it would open up chrome and display the 24h bandwidth graph on my TV.

Since version 2.8 (and also verified in 2.9), this gives the error below. This happens in chrome, firefox, all browsers I tried.

Can this be fixed?  (Or: is there another way to link directly to the bandwidth page and pass the username in the link itself?)

Thanks

___________________________________

ERROR occured!
There has been error while loading a page, please review debug data bellow if this is isolated issue.
If this is not an isolated issue, please create issue with details bellow and how to reproduce the error at https://github.com/Jackysi/advancedtomato2/issues.

Detailed information:
showHours@http://my_username:my_password@192.168.1.1/js/jquery.min.js line 2 > eval:29:1
init@http://my_username:my_password@192.168.1.1/js/jquery.min.js line 2 > eval:178:1
@http://my_username:my_password@192.168.1.1/js/jquery.min.js line 2 > eval:1:1
.globalEval@http://my_username:my_password@192.168.1.1/js/jquery.min.js:2:4243
.domManip@http://my_username:my_password@192.168.1.1/js/jquery.min.js:3:21389
.append@http://my_username:my_password@192.168.1.1/js/jquery.min.js:3:18980
.html/<@http://my_username:my_password@192.168.1.1/js/jquery.min.js:3:20344
.access@http://my_username:my_password@192.168.1.1/js/jquery.min.js:2:5778
.html@http://my_username:my_password@192.168.1.1/js/jquery.min.js:3:20018
loadPage/TomatoAJAX.onCompleted@http://my_username:my_password@192.168.1.1/js/advancedtomato.js:231:3
XmlHttp.prototype.onReadyStateChange@http://my_username:my_password@192.168.1.1/tomato.js:1603:1
THIS/<@http://my_username:my_password@192.168.1.1/tomato.js:1855:28
showSelectedOption is not defined

Jackysi, 2015-09-21:
I have tested for this issue and its happening on my router as well. It does however seem to be different issue from what you experience so I believe this problem is related to something else.

I will research and attempt to fix for the upcoming updates! Thanks for your report!

Jackysi, 2015-12-07:
Can you verify this issue is still there in 3.0-132? I believe I was able to fix it but I am not sure. I did test http://admin:password@192.168.1.10/#bwm-realtime.asp and it seems to work just fine.

Jackysi, 2016-01-08:
Fixed in 3.0-132.

closed 2016-01-08 by Jackysi