Jackysi/advancedtomato #75: D-Link DIR-868L RevB, Tomato build 131 - no 5Ghz Support?

RotateAt60MPH, 2015-10-02:
Hi All,

I just flashed tomato-DIR868L-AT-ARM-2.9-131-special.trx onto my DIR-868L RevB hardware.  The only way I could get this done successfully is from the emergency utility in the D-Link.  Ok, so far, so good.

However, when I booted the router, I only have 2.4Ghz.  I recall that's a problem with some LinkSys routers as well with Tomato.  Sometimes, you had to screw with the USB settings, but I don't know if that's the case here.

Is that the same case here?

Sigh - back to stock firmware again.  :P

Thanks for any advice!
Corey.

Jackysi, 2015-10-02:
Sadly you posted the issue into wrong section. I only support AdvancedTomato which is Tomato GUI modification. Original Tomato issues should be reported to Shibby who develops the firmware. This issue is related to Tomato not AdvancedTomato so I am closing it.

closed 2015-10-02 by Jackysi

RotateAt60MPH, 2015-10-02:
Thanks.  Will do.  Apologies.

Jackysi, 2015-10-02:
No need to apologize :) and thanks for understanding! My work is based on http://tomato.groov.pl/