Jackysi/advancedtomato #76: web ui totally f..ed up

zaprenki, 2015-10-04:
Hey! First and foremost I would like to congratulate you for your fantastic work with advanced tomato. Now to the point: I've been using your firmware in a very long time with my router (Asus RT-AC68U), never had an issue before. Then, for about a week ago I tried logging in to the web interface, only to realize that I couldn't. The page would just not load, giving me all kinds of errors. Otherwise the router worked perfectly, I was connected to the net, WiFi working, could telnet, no probs att all there.

I've tried everything possible, resetting, 30-30-30 resetting, stopping and starting httpd, nvram erasing, installing Asus's original firmware then reinstalling your's, then again, yeah, I've been testing everything I could. But nope, no WebUI. I'm usually using Chrome and all I got was those screens:
[image]
or
[image]
or in the best case
[image]

I tried with IE too, getting the exact same screens, even though from time to time I get this screen
[image]

Klicking the links in the menu makes no difference at all.

I'm really out of ideas here, any help would be welcome....

Thanks in advance!

zaprenki, 2015-10-04:
By the way I can telnet into the router, so if you need some info from there I can fetch it for you.

Jackysi, 2015-10-04:
Actually can you please open Chrome Console (F12, Developers mode) and see under "Console" tab what happens? (requires refresh f5 while console is running). This may tell us more about the problem but I really have no idea what that could be. Also perhaps if you would allow me to use Teamviewer I may be able to debug the problem easier. Not necessary though, since it's sunday I really don't have time today anyway.

zaprenki, 2015-10-04:
Thanks for quick reply. I tried with Chrome, after a lot of page refreshes i got this:

[image]

Did same thing with IE, got this instead:

[image]

You may use Teamviewer when you have some free time, email me so we can schedule this..

Jackysi, 2015-10-04:
Looks like JQuery is missing, what did you change since last time accessing GUI? I mean like did you install a adblocker, changed hosts file, changed network settings anything like that? Perhaps did you try to access GUI via different computer? Most issues like this are related to client not server (router in this case). I can check now in few minutes otherwise tomorrow sometime. My email is jacky@prahec.com

zaprenki, 2015-10-04:
Actually I didn't change a single thing in my computer since I last logged inte successfully in webui. Now that I tried logging in from my laptop, it works totally ok. I'm sorry for having wasted your time. A last question only: why would JQuery be missing when I didn't change a thing in my computer, and is there a way to get it back.

Thanks again for your answers ;)

Jackysi, 2015-10-04:
I honestly have no idea. Under "Network" tab in developers console you can see which resource is not loading properly and check maybe what is blocking it. However I would say an antivirus or something like that is blocking it. I really don't know out of my head, never seen this happen on a whole device. Maybe a browser but you have issue on all browsers so it's very strange. A virus could do this too but again never heard of one that would only block single resource heh

closed 2015-10-10 by Jackysi