Jackysi/advancedtomato #77: Web UI errors

spelaben, 2015-10-09:
Hi, I just switched to AdvancedTomato 2.9-131 K26AC USB AIO-64K using an ASUS AC66U. When I click on Basic Settings - Network the following error shows:

[image]

I tried this from 3 computers, on this one I'm using Chrome OS so there is no AV blocking anything. Also disabled AD blockers and tried to call this part of the UI from an incognito tab where all add-ons are disabled. It doesn't work. 

Anything I can do to help you troubleshoot the issue? Just let me know. :)

Jackysi, 2015-10-10:
To be honest these errors don't really tell me anything. I have been trying to find a way to debug UI issues for long time and the screenshot above was one of its attempts. I found a better way to debug it now but it is going to take a update to change the behaviour. Problem is original Tomato AJAX function attempts to catch exceptions which are then stopped in the script itself instead of displayed to browser. Since script can't handle exceptions as well as browser ( to show what actually went wrong ) the data is useless.

spelaben, 2015-10-10:
Nevermind, I erased NVRAM and started from scratch. This time all menus are showing. :)

Thanks for the information though.

Jackysi, 2015-10-10:
Oh, happy to hear that! 

Well in future if you hit an error (versions 3.0+) don't hesitate to check browser console log (F12) which will be far more detailed and will help solving issues easier. It is important to clear NVRAM on upgrades though =)

closed 2015-10-10 by Jackysi