Jackysi/advancedtomato #84: 2.4Ghz band not function on RT-AC56U past Tomato by Shibby release 123 and all Advanced tomato Releases

harveyit, 2015-11-03:
I recently acquired an ASUS RT-AC56U router in hopes to run a version of tomato.  I tried several releases of Shibby's tomato and Advance Tomato.   I flashed it using the ASUS firmware restoration tool and cleared the NVRAM after each fresh firmware flash.  However, only the 5GHz band would function.  On the web GUI it would show enabled but could not see it from any device.

On a hunch I started trying older versions.  I tried all versions of advanced tomato with the same result.  I tried every version of Shibby's tomato until i got to release 123 (tomato-RT-AC56U-ARM--123-AIO-64K.trx).  This release is working fine but id much rather have a newer release with bug fixes and improvement.  

Any assistance would be appreciated.

Jackysi, 2015-11-03:
This is not a support section. AdvancedTomato is only fork of Tomato by Shibby with re-designed graphical interface. These sort of reports are invalid and closed.

closed 2015-11-03 by Jackysi

harveyit, 2015-11-03:
Sorry about that.  On the Advanced Tomato site it says "If you encounter errors or issues with AdvancedTomato it's best that you report them on Github repository under Issues tab."  So thats what i thought I did.

Could you point me in the right direction.

Jackysi, 2015-11-03:
It's not a problem. I'm used to getting reports regarding firmware issues all though I'm not the one developing firmware. Check out Shibby's web site or try http://linksysinfo.org/ if you post a topic regarding your issue I am most cartain someone will help you out!

harveyit, 2015-11-03:
I appreciate it Jackysi.  Cheers