Jackysi/advancedtomato #85: Starting/stopping VPN client redirects to https://x.x.x.x/service.cgi#vpn-client.asp instead of https://x.x.x.x/#vpn-client.asp

mdPlusPlus, 2015-11-08:
Starting/stopping VPN client redirects to https://x.x.x.x/service.cgi#vpn-client.asp instead of https://x.x.x.x/#vpn-client.asp which doesn't give a 404 error but a "Secured connection failed" warning under Firefox and a ERR_EMPTY_RESPONSE error under Chromium.

hasafly, 2015-11-08:
I confirm this.

Jackysi, 2015-11-17:
Srry for late response, I will check it out and fix it in next release

jontis, 2015-11-21:
I was looking into this issue and understand you already have the solution.
May I suggest an enhancement related to this fix:
It would be great to be able to put in a friendly nickname for your vpn if you have two.

tparvu, 2015-12-28:
Decided I should search the bugs before opening a new one. :)

This is the same as the issue #105 and #95 

Jontis makes a good point, it would be nice to be able to assign a friendly nickname for the two configurations.

closed 2016-01-08 by Jackysi

mdPlusPlus, 2016-01-14:
May I ask why this has been closed? I can't find a commit which would fix the issue.