Jackysi/advancedtomato #90: Logs flooded with miniupnpd warnings

alindt, 2015-11-19:
On AdvancedTomato 3.0-132 K26ARM USB AIO-64K (and possibly others):

`Nov 19 13:38:25 daemon.warn miniupnpd[1510]: SendSSDPNotify(): truncated output (513>=512)`

Please consider shortening the OS_VERSION because the SSDPNotify packets seem to be longer than 512 and miniupnpd has a problem with that:

http://www.linksysinfo.org/index.php?threads/tomato-shibbys-releases.33858/page-53#post-262636
https://github.com/miniupnp/miniupnp/pull/129
https://github.com/miniupnp/miniupnp/commit/d002502e68d7be379ea6cae6f03d149577c2e19c

Jackysi, 2015-11-19:
http://www.linksysinfo.org/index.php?threads/tomato-shibbys-releases.33858/page-58#post-267361

closed 2015-11-19 by Jackysi

alindt, 2015-11-19:
If AdvancedTomato had used the unmodified Tomato version tag then I would understand that as a valid reason to dismiss this issue as "firmware bug, upstream's problem", but the version tag is under your control too since you modify it to add AdvancedTomato specific versioning.

https://bitbucket.org/jackysi/advancedtomato-arm/src/3636f5081bcf173f3c5f439e38c20ed1d4cea45e/release/src-rt-6.x.4708/router/shared/tomato_version?at=master&fileviewer=file-view-default

I've just tested Shibby's version 132 on a different router and it doesn't spam the logs.

Shibby's v132 version tag:
`Tomato v1.28.0000 MIPSR2-132 K26 USB Big-VPN`

AdvancedTomato's 3.0-132 version tag:
`Tomato v1.28.0000 -3.0-132 K26ARM USB AIO-64K`

There's at least one character that makes the tag longer for no reason (the first "-" from "-3.0-132).
As I see it, there's no change that upstream can commit in order to fix this issue on AdvancedTomato since you're the one that controls the version tag on your builds.

alindt, 2015-11-19:
The ".0000 -" part is completely useless, "Tomato v1.28 3.0-132 K26ARM USB AIO-64K" is more than enough and it's clear that it's not a separate project from TomatoUSB/Shibby's/etc since you explicitly state everywhere that your project is identical to Shibby's except for the GUI.

Just my 2 cents...

Jackysi, 2015-11-19:
I will check versioning and go for something as you suggested, Tomato v1.28 3.0-132 K26ARM USB AIO-64K