Jackysi/advancedtomato #95: "No data received" after stopping Open VPN Client

orange-buffalo, 2015-11-30:
#### Version
* 1.28.0000 -3.0-132 K26ARM USB AIO-64K, Asus RT-AC56U

#### Prerequisites:
* Running Open VPN Client

#### Test Case
* Go to VPN -> Open VPN Client -> Running Client
* Click "Stop"

ER: Page is updated with new client state
AR: HTTP error "No data received"

Jackysi, 2015-12-14:
I was unable to replicate this issue, but I will leave it open for few more weeks to see of anyone else can confirm it.

tparvu, 2015-12-28:
Oops, just noticed this issue, I just opened issue #105 I'll leave it up as I have a screenshot in there. I think the URL should be "/#vpn-client.asp" instead of "/service.cgi#vpn-client.asp" but that is just a guess on my part.

Jackysi, 2016-01-08:
Okay this has been fixed in GUI repository: https://github.com/Jackysi/advancedtomato-gui/commit/651d6f4369fa1f3d6f7260476d7c04976caf4a3f 
When releasing next version this fix will be included in main GUI.

closed 2016-01-08 by Jackysi

orange-buffalo, 2016-01-16:
Great, thank you!