Jackysi/advancedtomato #100: Add CSS skin to Adminer.

NoirHD, 2015-12-10:
This is fairly simple to do, just put the file adminer.css alongside adminer.php.

[image]

Jackysi, 2015-12-11:
I don't know how many people actually use adminer. I use this skin too but not on my router though, we'll see if this is possible to embedded into Tomato, problem may be additional size (already very close to limit at most non-ARM routers).

NoirHD, 2015-12-14:
Could you just do it for ARM then? I own an R7000. I also sent you a message some time ago about advancedtomato on Linksys E900 but your contact form doesn't seem to be working properly. I would greatly appreciate it. Thanks!

Jackysi, 2015-12-14:
I receive daily emails that make no sense or answers are already available online, such emails I ignore. There is also many contact messages where people ask for support which I don't offer =/.

I will see but generally I don't do differences between ARM/RT-AC. My goal is to have united interface for easier maintenance and so on. I might however lose all other routers in future and only stay with ARM.

NoirHD, 2015-12-14:
In your FAQ you said that if Shibby has builds available for my device (which he does) you might consider adding an advancdtomato build aswell. It means a lot to me, thank you!

Jackysi, 2015-12-14:
Yes but only RT-N / RT-AC / ARM builds, the 900 is RT.

NoirHD, 2016-01-06:
Actually, E900 is K26RT-N according to Shibby's router list page. It should be the same FW as ASUS RT-N10U.

Jackysi, 2016-01-20:
https://github.com/Jackysi/advancedtomato-gui/commit/96b4ed3f9dd6720dc9eb8e4fc5abd5696b5931ac

closed 2016-01-20 by Jackysi