Jackysi/advancedtomato #106: what's wrong with my MAKE?

thinktip, 2016-01-02:
I am an AdvancedTomato fans,  Thank you for your good job! I very like your project!
I complied the AdvancedTomato by myself use your project
https://github.com/Jackysi/advancedtomato-gui
&
https://github.com/Jackysi/advancedtomato

I copy all the gui files to router/www, and chmod  remcoms2.sh & dnscrypt-helper.sh 0777 & +X, then 'make wndr4500v2z', it works and I flashed the rom successful, but when I login the GUI, it only show the header, like the pic blow:
http://www.thinktip.com/temp/adterr.jpg

Sorry for my poor English, wish you can give me some advice.

Jackysi, 2016-01-02:
Check browser console, though in General you don't have to copy or change anything when using https://github.com/Jackysi/advancedtomato. Everything is already there.

thinktip, 2016-01-02:
Use advancedtomato defaul www is ok, but I want translate the www to Chinese, what should I do?

there is a error in console:
"Uncaught TypeError: Cannot read property '2' of null" status-overview.asp:36 

the line is:
"gui_version = gui_version.match(/^1.28\.0000 (MIPSR2\-|\-)?(.*)/)[2] || ''; "

Jackysi, 2016-01-02:
The above error is issued when image name is not matching regex. when compiling you need to add V1 and V2 variables. I honestly have no idea how are you going to translate the interface. However this is not a support section, for support please check out https://linksysinfo.org/ and ask other authors there. For the issue above use V1 and V2 variables something like
˙˙˙
make IMG V1=RT-N5x V2=-3.0-132
```

closed 2016-01-02 by Jackysi

thinktip, 2016-01-02:
Thank you very much, let me try.