Jackysi/advancedtomato #109: ASUS RT-N10U enters recovery automatically on boot after 3.0-132 FLASH.

NoirHD, 2016-01-06:
Hello,

There seems to be a problem with your latest 3.0-132 build for ASUS RT-N10U. It gets stuck into recovery after flash (I have tried multiple times, TFTP included) and I got the same result. Everything works fine with 2.9-131. I will try to update from 2.9-131 and tell you the result later. I will report this to Shibby as well but I cannot find a way to contact him on his website. Thank you for your work!

NoirHD, 2016-01-06:
I tried to upgrade from 2.9-131 and I get the following error: File is too big to fit in MTD.

Jackysi, 2016-01-08:
Will be checked, but this issue tracker is meant for GUI issues and not firmware images problems... One is for "BUGS" and one is for "Problems" / "Support". Please rather email me instead in future...

closed 2016-01-08 by Jackysi

NoirHD, 2016-01-21:
Thanks for the reply, I surely will. As a side note: I know you are busy building your GUI for all models these days and it takes time so I might be able to take some burden away from you. If you could provide some steps I should follow, I will gladly build those images for you (I possess both Windows and Linux systems on physical machines highly capable in terms of raw CPU power).

NoirHD, 2016-01-21:
Thanks for the reply, I surely will.