Jackysi/advancedtomato #110: BitTorrent Client - slow upload

jurcacristian, 2016-01-19:
Hi guys, 

Your router brand & model : Asus RT-N18U
Interface version: Tomato Firmware 1.28.0000 -3.0-132 K26ARM USB AIO-64K
Screen shot or error details : Slow upload speed on BitTorrent Client
How to reproduce the issue/error : Well, i'm trying to use the drive connected in USB port as a media share/DLNA server . Everything works more that fine, except one thing : the upload when using the builtin Bittorrent client is extremly slow. I've had the same issue when using the asus stock firmware and i had to activate QoS and set the bandwidth values from ISP (500Mbps/200Mbps)

The behavior now is : i'm downloading with acceptable speeds up to 10MB and there is no upload at all . After the download is finish i can see small traces of upload 14~344 kbps.  I've downloaded the same same movie from my desktop and it's working in the same way (the QoS was stopped)

I've tried the following fixes:

- i've changed the settings activated QoS adn input :"Inbound-Outbound Rates / Limits" from ISP
- i've tried tweaking the Bitorrent Client by inforcing same limits from application
- i've added no limits for Bandwidth Limiter for 
Can someone please help me out on what i'm doing wrong ?

Cheers,
Cristi.

Jackysi, 2016-01-20:
This problem is related to router firmware, please report it to Shibby https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open! This is issue tracker for GUI related issues.

closed 2016-01-20 by Jackysi

jurcacristian, 2016-01-20:
Ok, many thanks for your reply.
Cristi