Jackysi/advancedtomato #112: Real Time Bandwidth monitor visualization bug

Desani, 2016-01-27:
[image]

If you let the Real-Time bandwidth monitor reach the end of the graph there is a line that jumps from the to the beginning spanning the entire graph. I also checked the Real-Time IP monitor and it suffers from the same issue. See Below.

[image]

Jackysi, 2016-03-04:
I haven't noticed this issue. How did you manage to start it so I can replicate and fix it?

Desani, 2016-03-09:
All you have to do to reproduce the issue is visit any of the real time graphs and let the visualization get to the end of the graph. So sit on the page for about a minute or two and when it reaches the end of the graph you get the visualization as pictured above.

mzhboy, 2016-03-10:
[image]

AdziePahlevi, 2016-05-23:
The bug still exist on 1-136 K26 USB AIO version. What you can do to reproduce the issue is, just wait until 10 minutes (because you have to get the graph from right to the left).. 

[image]

Jackysi, 2016-05-23:
there has been no UI changes for last 3 shibby updates. Its still version 3.1 so I don't know why would u expect a fix?

AdziePahlevi, 2016-05-23:
Hi Jackysi, okay I thought there was some update. I just recently using advancetomato, anyway this is minor bug not affecting the performance nor functionality..

Jackysi, 2016-05-24:
its okay, well there was Shibby update but I only modified his updates and not GUI it self. I just have no time =/

closed 2016-10-02 by Jackysi