Jackysi/advancedtomato #115: Redirect through VPN not working?

aktiv-d26, 2016-01-31:
Hi, 
I have a problem using the 'redirect thorugh vpn' option. After successfully connecting to an open vpn, all ip's are going through vpn, even when I check the 'redirect through vpn' option under the 'routing policy' tab. For example, when I enable 'From source IP' to 192.168.1.2, still all IP's are behind the vpn. There's no ISP IP connection possible.... What could be the problem here?

Please note that the WAN IP differs from the LAN IP because the advanced tomato is used as a second router. WAN IP and gateway is 192.168.0.X and LAN IP is 192.168.1.X but in my humble opinion that shouldn't be a problem for redirecting the vpn. 

Greetings

Jackysi, 2016-02-05:
This problem is related to Tomato router firmware, please report it to Shibby https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open! This is issue tracker for GUI related issues.

closed 2016-02-05 by Jackysi

xhaloz, 2017-03-30:
I have the same issue.  Did this ever get fixed?