Jackysi/advancedtomato #117: Needed to add iptables rule to make external access possible in router mode

nobodypb, 2016-02-02:
In router mode the httpd service doesn't bind to the external ip address. For getting access by using interernal ip however I need to set a filter rule.
iptables -A INPUT -d 192.168.180.1 -p tcp -m tcp --dport 80 -j ACCEPT
Where 192.168.180.1 is the internal ip.

Jackysi, 2016-02-05:
I am sorry but this bug is related to Tomato it self and not AdvancedTomato Graphical User Interface. I know its hard to understand but this project is fork of Tomato by Shibby. This project is only Graphical User Interface change (which is actually the hardest part) but the core functionality is unchanged. So for support and issues reports on Firmware/Linux/Core please contact Shibby. I do not have enough knowledge to poke around Firmware so I rather not do that.

closed 2016-02-05 by Jackysi

nobodypb, 2016-02-07:
I know this isn't directly related to GUI but I wrote it here for others to notice if they stumble across the same issue. Also I didn't know you don't modify the core at all. I'll report on Shibbiys bug tracker.
By the way, thank you for providing such a nice GUI for tomato.

dihego, 2016-02-07:
Jacky,

A few from work lived the gui and features. No I haven't modified the core at all.  

If you were to charge for it I would pay for this type of GUI - of course as long all features work  as they should ;)

Thanks,
Diego