Jackysi/advancedtomato #121: Wrong port order on VLAN page for R7000

nobodypb, 2016-02-07:
So here is something that actually concerns GUI. 
In advanced-vlan.asp
```
if (nvram['boardrev'] == '0x1301')  { //Netgear R7000
COL_P0N = '4';
COL_P1N = '3';
COL_P2N = '2';
COL_P3N = '1';
COL_P4N = '0';
break;
}
```
Is set. That appears to be wrong for my r7000. Correct for me seems to be
```
if (nvram['boardrev'] == '0x1301') { //Netgear R7000
COL_P0N = '1';
COL_P1N = '2';
COL_P2N = '3';
COL_P3N = '4';
COL_P4N = '0';
break;
}
```

Some information about my model:
```
# nvram show | grep model
t_model_name=Netgear R7000
t_model=84

# nvram show | grep board
boardrev=0x1301
pci/2/1/boardflags=0x30000000
boardtype=0x0665
pci/1/1/boardflags=0x1000
boardflags2=0x00000000
board_id=U12H270T00_NETGEAR
pci/1/1/boardnum=57359
pci/1/1/boardrev=0x1150
pci/1/1/boardtype=0x661
pci/1/1/boardflags2=0x100002
pci/1/1/boardflags3=0x10000003
pci/2/1/boardtype=0x621
boardflags=0x00000110
pci/2/1/boardnum=20507
pci/2/1/boardrev=0x1451
pci/2/1/boardflags2=0x300002
pci/2/1/boardflags3=0x10000000
boardnum=32
```

aerickson, 2016-02-16:
Already fixed upstream? 

https://bitbucket.org/pl_shibby/tomato-arm/commits/57e7c67c121e88bb04425d5605db2a0a3e237bf4

Jackysi, 2016-05-29:
From my knowledge this has been fixed with few releases now ^^ thanks for report!

closed 2016-05-29 by Jackysi