Jackysi/advancedtomato #125: Router Policy

Pascall1985, 2016-03-04:
Hello im recently buy a netgear r7000 and use PIA openvpn. But with ddwrt i can use selective VPN witin the router policy. But when i use it within Tomota everthing still going trough the vpn. I want to have some ip through vpn some through regulary ISP. What do i wrong. I used the right tab. From source ip and than for example. 192.168.1.2/32  what works in ddwrt. 

kind regards

Jackysi, 2016-03-04:
This is not a support section!

closed 2016-03-04 by Jackysi

Pascall1985, 2016-03-04:
You design it. And it wont work. Where can i Find the support forum?

Date: Fri, 4 Mar 2016 02:35:53 -0800

Closed #125.

Jackysi, 2016-03-04:
I did not design these functions. They're part of Tomato. AdvancedTomato is a FORK (base) of Tomato project by Shibby and so the modifications are only visual and not functionality based. With that in mind I did not create, modify or delete any functions from Tomato. 

http://www.linksysinfo.org/ is main Tomato forum.