Jackysi/advancedtomato #131: MAC Issue

Aleks-Z, 2016-03-17:
In last production build (1.34) i have mac problem. I cann't change any mac address. Check out

NETGEAR WNR3500lv2

Jackysi, 2016-03-17:
I hope you are kidding with that attitude... I'm aware of the problem....

devz3ro, 2016-03-18:
I apologize for sending in a bug request on the website. Looks like you're already on top of it, thanks - appreciate your time and effort.

santomet, 2016-03-18:
Is it going to be fixed anytime soon? Or it is better to downgrade for now, because i really need this feature :(

Jackysi, 2016-03-18:
Issues are fixed through versions. I don't have the time to actually fix everything as soon as its reported. I did fix the problem but the fix is only available in source code. If you wish you can compile firmware on your own and get the fix now, otherwise waiting for next release is a must. The update was released just yesterday and I'm not going to re-compile every thing today again (takes 36 hours to compile firmware for all routers).

closed 2016-03-18 by Jackysi

santomet, 2016-03-19:
Yeah, sure, that's all i need. Thanks!

Aleks-Z, 2016-03-23:
Big thanks, @Jackysi . But i couldn't find tutorial for compile of GUI firmware . [Doesn't work.](https://github.com/Jackysi/advancedtomato2/blob/advancedtomato-ac/readme.md) Does exist it ?

Jackysi, 2016-03-23:
https://bitbucket.org/pl_shibby/tomato-arm/issues/15/compilation-libjson-c-fails and home page of repository https://github.com/Jackysi/advancedtomato