Jackysi/advancedtomato #132: No access to ipv6_dhcpd in web interface

npt, 2016-03-17:
The ipv6_dhcpd NVRAM setting isn't accessible, though I would think it would be accessible on the Advanced DHCP page the way ipv6_radvd is.

Jackysi, 2016-03-22:
U should talk with Shibby about that. I'm GUI developer and I don't work firmware features
http://www.linksysinfo.org/index.php?threads/tomato-shibbys-releases.33858/page-63

npt, 2016-03-22:
This is a GUI feature. The ipv6_dhcpd setting is already there in the firmware, and is present in the GUI in Shibby's Tomato, but not present in AdvancedTomato. Compare:
https://github.com/tomatofirmware/tomato/search?utf8=%E2%9C%93&q=ipv6_dhcpd
https://github.com/Jackysi/advancedtomato/search?utf8=%E2%9C%93&q=ipv6_dhcpd

Jackysi, 2016-03-22:
I did not notice this please accept my sincere apologies! I will merge your commit, hopefully that's all that is missing?

closed 2016-03-22 by Jackysi