Jackysi/advancedtomato #135: Turning on QOS disables internet access

kacheng, 2016-03-21:
I have an Asus RT-N16.

I just upgraded to AdvancedTomato Version 3.1-134.
I checked the md5sum of the firmware file before upgrading.
I cleared nvram on installation.

I recreated my wifi, vlan, ddns, adblock script, and openvpn client settings.
Openvpn client is not currently running (i.e. 'stopped').

Next on the list is to reactivate QOS. 
I have not made any changes, but as soon as I 'Enable QOS', internet access dies.
All classifications on the QOS Graphs page show as 'unclassified'.
Unchecking 'Enable QOS' restores internet access.

Anyone else experiencing this?
Any ideas on how to troubleshoot this?
Thanks!

Jackysi, 2016-03-22:
http://www.linksysinfo.org/index.php?threads/tomato-shibbys-releases.33858/page-63

closed 2016-03-22 by Jackysi

kacheng, 2016-03-23:
Thanks. Sounds like I will revert back to Version 3.1-132 to get QOS working.