Jackysi/advancedtomato #136: graphical interface displays nothing

CoderRunner, 2016-03-22:
After I flash the R6300v2-AT-ARM-3.1-134-AIO-64K.trx
my graphical interface becomes this..

Now, I don't know how to flash my router again..

[image]

CoderRunner, 2016-03-22:
I've solved it.
It may result from not erasing nvram before I flash R6300v2-AT-ARM-3.1-134-AIO-64K.trx.

The solution is 
1. ssh/telnet login router
2. enter nvram_erase
3. enter reboot

Thanks.

Jackysi, 2016-03-22:
You can also use http://192.168.1.1/upgrade-plain.asp which doesn't load UI at all. Its simple ASP to upload another image when main UI fails. I can't debug the problem you encounter without any details from console log (F12).

closed 2016-03-26 by Jackysi