Jackysi/advancedtomato #139: IP traffic, 24 hours view is broken

aditaga, 2016-03-24:
The IP Traffic, last 24 hours graph is broken on my router. I have a Netgear WNR3500L-v2.

IP traffic data is not displayed correctly and the clickable IP address tabs at the top of the graph are only [visible](http://s30.postimg.org/y3kxbrtpd/bwm_ipt_24_1.png) for a couple of seconds, after which they [disappear](http://s30.postimg.org/6hi5r3ach/bwm_ipt_24_2.png).

Another odd thing is the last item on the "IPs currently on graphic" [drop down list](http://s30.postimg.org/6rpoafoyp/bwm_ipt_24_IPs_currently_on_graphic.png).

Jackysi, 2016-03-26:
Please use F12 developers tools to copy/paste actual problem.

aditaga, 2016-04-02:
Hi, the console shows no errors.
[image]

aditaga, 2016-04-11:
Hi, I found the problem. It was on my side. 

The problem was that I had enabled saving of IP traffic monitoring history to a file on disk, but the file got corrupted, it seems. I cleared the bad data by using the 'Create New File' option and now IP monitoring works again.

Jackysi, 2016-05-30:
Oh, I referenced wrong issue. agh well this one is for closing thanks for response! And sorry im so damn late! :/

closed 2016-05-30 by Jackysi