Jackysi/advancedtomato #140: Getting this error pretty regularly

shaneypoopers, 2016-03-24:
ReferenceError: stats is not defined
    at eval (eval at  (http://192.168.0.1/js/jquery.min.js:2:4243), :5:27)
    at Function.x.extend.globalEval (http://192.168.0.1/js/jquery.min.js:2:4243)
    at x.fn.extend.domManip (http://192.168.0.1/js/jquery.min.js:3:21391)
    at x.fn.extend.append (http://192.168.0.1/js/jquery.min.js:3:18985)
    at . (http://192.168.0.1/js/jquery.min.js:3:20357)
    at Function.x.extend.access (http://192.168.0.1/js/jquery.min.js:2:5780)
    at x.fn.extend.html (http://192.168.0.1/js/jquery.min.js:3:20020)
    at Object.TomatoAJAX.onCompleted (http://192.168.0.1/js/advancedtomato.js:231:29)
    at Object.XmlHttp.onReadyStateChange (http://192.168.0.1/tomato.js:1603:6)
    at XMLHttpRequest. (http://192.168.0.1/tomato.js:1855:33)stats is not defined

It seemed to start happening after I tried to enable IPv6, my modem uses DS-Lite and I am not sure it is compatible. I disable the IPv6 but still getting this error.

Jackysi, 2016-03-26:
I doubt this problem is related to the GUI it self, I'm pretty sure its something with Tomato IPv6. I'll give it a look in few days.

Jackysi, 2016-10-02:
Verified to be an IPv6 issue with some devices / builds. Should be fixed with 137+

closed 2016-10-02 by Jackysi