Jackysi/advancedtomato #141: The field "qosl_ulr" is invalid. Please report this problem.

ionas, 2016-03-24:
Helloes,

I wanted to limit my wireless guest vlan to QoS = junk while the wireless internal and hardwired internal network have best performance always, so that if someone joins my office and I hand over some wifi credentials he cannot screw up my internet/network performance:

[image]
[image]
[image]
[image]
[image]

However... I cannot save that setting.

**Router:** Asus RT-N18U
**FW:** 1.28.0000 -3.1-135 K26ARM USB AIO-64K

Jackysi, 2016-03-26:
135 is beta/experimental. I'll look into it with next update

ionas, 2016-03-26:
How do I know from the Downloads page if something is considered "stable" or beta/experimental

Jackysi, 2016-03-27:
You don't. Warning was added because Shibby asked me to warn users since versions 132+ are unstable for now. MultiWAN features are hard to implement.

ionas, 2016-03-27:
* Thus if I don't need MultiWAN 131 latest is the most stable and useful one?
* Could you add "BETA" as a section under downloads in any way?

Jackysi, 2016-03-27:
The warning is all you need, its totally explanatory. It says that versions 133 and above are currently not considered stable. At this time I won't bother and waste time on this problem while facing many others on the firmware. Remember this is free, open source project.

closed 2016-10-19 by Jackysi

gaia, 2017-03-22:
I am getting the same issue on 138. @Jackysi is your Mar 27 2016 comment still valid?