Jackysi/advancedtomato #143: QoS: All (some?) traffic using default class limits?

Mantriur, 2016-03-26:
Not sure if this is a bug or if I am missing something obvious.

Advanced Tomato already includes a QoS rule setting NNTP to FileXfer. I added another one for Steam downloads. When I look at the connection details, the traffic is correctly classified. However, both Steam and NNTP peak at roughly 2MB/s, both individually and simultaneously. QoS is configured to 70mbit. When I change the default class from P2P/Bulk to FileXfer, they correctly download at over 5MB/s. It seems not the configured and reported class is applied, but the default one.

This is on an Asus RT-AC68u running latest Advanced Tomato 1.28.0000 -3.1-135 K26ARM USB AIO-64K.

Jackysi, 2016-03-26:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a  GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

This is an issue tracker for AdvancedTomato GUI related issues!

closed 2016-03-26 by Jackysi

Mantriur, 2016-03-26:
Whoops, sorry! Thanks for the link!