Jackysi/advancedtomato #144: qod missing some options

lceron80, 2016-03-27:
The last two updates 134 and 135 have qdisc scheduler options missing.  Before, the two mentioned updates I'd have four qdisc scheduler options which include: sfq, pfifo, codel, and fq_codel.  The last two updates only have sfq and pfifo.  Is there are reason why the last two updates removed the codel and fq_codel options?  In my OCD I feel that the fq_codel gives me better performance vs the sfq and pfifo options.  Would it be too much trouble to bring the codel and fq_codel options back?  Thank you.

Linksys EA6500V2
Advanced Tomato 3.1-135

Jackysi, 2016-03-27:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

lceron80, 2016-03-27:
yep, my understanding was that AT is a fancy GUI for Tomato Shibby.  I have reloaded Shibby's version and Shibby has the four qdisc scheduler options instead of only two by AT.  Take a look at my two videos with Shibby and AT loaded.

AT https://mega.nz/#!GBVDRLRQ
Shibby https://mega.nz/#!3FkmBS6L

Westacular, 2016-05-23:
This _is_ an issue with the AdvancedTomato GUI, not the underlying firmware.

If you look at the history for `qos-settings.asp`:
https://bitbucket.org/jackysi/advancedtomato-arm/history-node/365731748f02/release/src-rt-6.x.4708/router/www/qos-settings.asp?at=master

you can see Kyle Sanderson's commits to add `codel` and `fq_codel` support. The underlying changes from those commits (e.g. in `qos.c`) are still present, however in the latest commit for that file (`3657317 Rebased History (FIRST COMMIT)`) the UI options for selecting them were inadvertently clobbered. What was

```
{ title: 'Qdisc Scheduler', name: 'qos_pfifo', type: 'select', options: [['0','sfq'],['1','pfifo'],['2','codel'],['3','fq_codel']], value: nvram.qos_pfifo }
```
became
```
{ title: 'Qdisc Scheduler', name: 'qos_pfifo', type: 'select', options: [['0','sfq'],['1','pfifo']], value: nvram.qos_pfifo }
```

closed 2016-06-09 by Jackysi