Jackysi/advancedtomato #145: Wan ip is locked on renewing

sstraus, 2016-03-27:
With the latest version wan ip keep reporting as renewing but the ip is already assigned and available.

Regards,

S.

Jackysi, 2016-03-27:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-03-27 by Jackysi

sreknob, 2018-03-14:
Just putting this here for reference to save future people some googling.
This is actually appears to be an issue with the default settings of leaving DHCP on when PPPoE is active.
Also affects WAN IP interface to always be 0.0.0.0 under DDNS settings.
Disable DHCP under WAN settings to resolve, fixed it for me.
Credit to user Chen Mulong here: https://bitbucket.org/pl_shibby/tomato-arm/issues/41/r7000-at-arm-31-135-aio-64k-port-fw-dmz